CadQuery/cadquery-freecad-workbench #135: show cqparts within FreeCad

tolot27, 2018-12-18:
Currently, cqparts can be shown using the `display` function and an external HTTP server and FreeCad 0.18 freezes afterwards. It would be great if the parts would be shown within FreeCad.

If I try this with the `show` or `show_object` function, I get:

```
Error executing CQGI-compliant script. 'Box' object has no attribute 'val'
```

An MWE is the Wrapping cube example from https://github.com/cqparts/cqparts

jmwright, 2018-12-18:
@tolot27 This workbench already supports/includes cqparts, but there's no example yet. https://github.com/jmwright/cadquery-freecad-module/issues/118

The [toy car example](https://raw.githubusercontent.com/cqparts/cqparts/master/examples/toy_car.py) works for me in FreeCAD 0.17 and the appimage of 0.18 for me. The only reason I can think of for cqparts to launch the web display is that the FreeCAD environment is not being detected properly. Can you post your FreeCAD version/lib information here?

Tagging @fragmuffin to make sure he's aware of this discussion.

tolot27, 2018-12-18:
FreeCAD 64-bit: 0.18.15417 (Git)
Python: 2.7.14
CadQuery: v1.2.0
cqparts: 0.2.2.dev1

I don't know how to find the version of the CadQuery-FreeCad-Module but I installed it at 15th of December. The file cadquery-freecad-module/__init__.py still contains version 1.0.

jmwright, 2018-12-18:
@tolot27 did you install this workbench from the FreeCAD add-ons manager? If so, you have the latest from master on Dec 15, which is what I'm running. If you installed cqparts via pip, that might be the difference. I'm using the version that's bundled with this workbench.

tolot27, 2018-12-18:
With FreeCAD 0.17.13541 (official release) and the same environment it works flawlessly.

jmwright, 2018-12-18:
Interesting. Thanks for testing that.

fragmuffin, 2018-12-19:
Thanks for tagging me @jmwright 
`cqparts` detects freecad as an environment for display from the presence of the `MYSCRIPT_DIR` environment variable. ([link to code](https://github.com/cqparts/cqparts/blob/master/src/cqparts/display/freecad.py#L13))

Has that changed?

fragmuffin, 2018-12-19:
oh!! it won't work if you have `show_object` anywhere else in the code ([`cadquery-freecad-module` code](https://github.com/jmwright/cadquery-freecad-module/blob/223bda63c9959995751819cb029fa9f857d57591/CQGui/Command.py#L228))
Even if it's not executed.

This is the `cqgi` _magic_ surrounding `show_object` that we're trying to remove (can't find the issue just now)

jmwright, 2018-12-29:
I can confirm what @fragmuffin is saying. I'm going to close this for now. @tolot27 feel free to re-open if the above didn't solve your problem.

closed 2018-12-29 by jmwright