CadQuery/cadquery-freecad-workbench #138: os.environ.get("MYSCRIPT_DIR") is None

asukiaaa, 2019-05-19:
I want use a path of directory on a script but `os.environ.get("MYSCRIPT_DIR")` returns None.
I tried to use `__file__` but it is not defined on FreeCADGui environment.
(I can use it by running a script with using `freecadcmd`.)

So I get the path from `getActiveCodePane` like this.
```py
import Shared
filepath = Shared.getActiveCodePane().get_path()
dirpath = os.path.dirname(os.path.abspath(filepath))
```
By referencing https://github.com/jmwright/cadquery-freecad-module/blob/59bbbc39580b9629ee839cb2ca95584f0d502ef7/CQGui/Command.py#L154 and https://github.com/jmwright/cadquery-freecad-module/blob/59bbbc39580b9629ee839cb2ca95584f0d502ef7/CQGui/Command.py#L228.

My Questions
- Is there any way to activate `os.environ.get("MYSCRIPT_DIR")`
- Is it no problem to use `getActiveCodePane` to get path on a script?

My environment
- OS: Ubuntu18.04
- FreeCAD: 1.8.1
- CadQuery: 1.2.0 (checked on python console on FreeCAD)

Thank you.

jmwright, 2019-05-19:
@asukiaaa I see that you've posted over on the CadQuery 2.0 repo, and Adam has mentioned CQ-editor to you. I would recommend moving to CQ 2.0 and CQ-editor since most of our effort is going to go into those from now on.

With that said, I'm not going to abandon this module, and a I'm surprised that the environment variable isn't working. I run Ubuntu 18.04, but don't use the environment variable very much. I'll take a look when I get a chance.

easyw, 2019-05-19:
> With that said, I'm not going to abandon this module

Yeah! This is a very useful module for FreeCAD and KiCAD :smile:
@jmwright thanks again for your work on this.

asukiaaa, 2019-05-19:
@jmwright 
> I'm not going to abandon this module
> I'll take a look when I get a chance.

Thank you.
I try CQ-editor and expect to resolve the issue.

jmwright, 2019-05-26:
@asukiaaa I am not able to reproduce the problem. The following code works fine on Ubuntu 18.04 when run using this module in FreeCAD 0.18.
```
import cadquery as cq
import os

print("ENV: " + os.environ['MYSCRIPT_DIR'])
```
It outputs what I would expect:
```
ENV: /home/jwright/Downloads
```
The only thing that I can find that's not quite right is that the environment variable is not refreshed after doing a Save-as on a script.

asukiaaa, 2019-05-27:
@jmwright 
Thank you for checking the vehavior.

I found that a script that has `show_object(` causes this problem even if `show_object(` is commented out.
[image]

I start FreeCAD and open and run the following script, the problem occurres.
```
import cadquery as cq
import os

dirpath = os.environ.get('MYSCRIPT_DIR')

box = cq.Workplane('XY').box(1,2,3)

# show_object(

filepath = dirpath + '/box.stl'
with open(filepath, 'w') as f:
	cq.exporters.exportShape(box, 'STL', f)
print('exported', filepath)
```

Error comment that cause on `filepath = dirpath + '/box.stl'`.
```
Error executing CQGI-compliant script. unsupported operand type(s) for +: 'NoneType' and 'str'
```

A script that has `show_object` without `(` seems no problem.
[image]

Once I succeeded in using `MYSCRIPT_DIR` without `show_object(`, the error was resolved.
[image]

Do you have any idea to resolve this?

jmwright, 2019-05-27:
@asukiaaa Thanks for the information. I've been able to reproduce this problem in Ubuntu 18.04 now. This script fails with the error you've posted in 18.04, but works fine in Windows 10.
```
import cadquery as cq
import os
import FreeCAD

dirpath = os.environ.get("MYSCRIPT_DIR")

#show_object(

box = cq.Workplane("XY").box(1, 2, 3)

filepath = dirpath + "/box.stl"
FreeCAD.Console.PrintMessage("exported" + filepath)

show_object(box)
```
FreeCAD uses a bundled version of Python in Windows and the system's version of Python in Linux. I'll probably start there to debug this.

asukiaaa, 2019-05-28:
@jmwright 
Thank you for checking.
I'm happy if I can use `MYSCRIPT_DIR` on Linux.

jmwright, 2019-05-28:
@asukiaaa Please update to the latest master either manually, or through FreeCAD's addons manager. The problem should be fixed now.

asukiaaa, 2019-05-29:
@jmwright 
I checked that my script succeeded in using `MYSCRIPT_DIR` with `show_object(`.
Thank you for updating.

closed 2019-05-29 by asukiaaa