CadQuery/cadquery-freecad-workbench #140: error on startup CQ_Logo.svg does not exists

tolot27, 2019-07-07:
If I start the Windows version of FreeCAD v0.19.xxxxx, I get the following error message:

[image]

This is caused by line https://github.com/jmwright/cadquery-freecad-module/blob/59bbbc39580b9629ee839cb2ca95584f0d502ef7/InitGui.py#L16

jmwright, 2019-07-07:
Thanks for reporting this. I'll take a look to see if they've changed the handling of icons, but 0.19 is unstable right now, so it could just be a temporary bug.

luzpaz, 2019-08-19:
Still persists  
[image]

luzpaz, 2019-08-19:
0.19 has a long release cycle...please consider fixing this before the next release.

luzpaz, 2019-09-26:
This bug makes FreeCAD irritating to use after a while

jmwright, 2019-09-26:
@luzpaz Is there any information about how icon handling has changed in 0.19? I looked around a little bit, but didn't see anything.

luzpaz, 2019-09-26:
I do not know.

luzpaz, 2019-09-27:
@jmwright perhaps https://forum.freecadweb.org/viewtopic.php?f=4&t=39660

easyw, 2019-09-27:
@jmwright 
you could use a different way to add the icon...
here a link to what I do for my wbs
https://github.com/easyw/Manipulator/blob/master/InitGui.py#L40
Just go from Line 40 to Line 56
I use an external file as a path locator for the folder, then I add the path for the icon directly as a svg file.

jmwright, 2019-09-27:
@easyw I'll give it a try, but I'm not sure how to know for sure when it's fixed. 0.19 out of the freecad-daily ppa doesn't throw an error for me.

easyw, 2019-09-27:
@jmwright 
I noticed this issue only on windows releases and only on not win Conda.
Everything is fine also for me on Linux.

easyw, 2019-10-03:
@jmwright 
you can also test the issue using i.e. this FC Appimage
https://github.com/FreeCAD/FreeCAD/releases/download/0.19_pre/FreeCAD_0.19-18353-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage

easyw, 2019-10-03:
@jmwright 
I think this PR should solve this issue
https://github.com/jmwright/cadquery-freecad-module/pull/143

jmwright, 2019-10-03:
@tolot27 @luzpaz Please update this workbench and make sure that this issue is fixed for you.

Big thanks for @easyw for the contribution to fix this.

luzpaz, 2019-10-03:
I can't reproduce anymore.

closed 2019-10-03 by jmwright