CadQuery/cadquery-freecad-workbench #142: How to write STEP file from Surf or volume use this cadquery package ?

selvakarna, 2019-10-03:
How to write STEP file from Surf or volume use this cadquery package ?

jmwright, 2019-10-03:
@selvakarna I'm not sure that CadQuery 1.0, which is what's bundled with this workbench, is capable of exporting a surface to a STEP file. If you want to export a volume, here is an [example from the tests](https://github.com/dcowden/cadquery/blob/master/tests/TestCadQuery.py#L84) that shows the syntax. You could try it with a surface, I'm just not sure if it will work properly.

You could always generate the model, then export it using FreeCAD's `File > Export` menu option.

The majority of the community has moved to [CadQuery 2.0](https://github.com/CadQuery/cadquery), and that's where most effort will be placed from now on. You might want to consider switching to it.

selvakarna, 2019-10-03:
Hi @jmwright , i don't wont menu option and all, i m asking supporting python scripts ?

jmwright, 2019-10-03:
@selvakarna The code below works for me.
```python
import cadquery as cq

shape = cq.Workplane("XY").box(1, 1, 1)

shape.val().exportStep("/home/jwright/Downloads/mymodel.step")
```

selvakarna, 2019-10-03:
Okay thank you @jmwright , i will update you ..........

selvakarna, 2019-10-03:
Hi @jmwright , when i import cadquery i got error :

Error:
 import FreeCAD
ImportError: dynamic module does not define module export function (PyInit_FreeCAD)

  so how to install FreeCAD? i have always installed freecad ? not like FreeCAD?

selvakarna, 2019-10-10:
windows also showing error?

can you share just commands :

how to install FreeCAD package in windows flatform ? you dont know?????????????//

jmwright, 2019-10-10:
@selvakarna I asked you questions on the second issue you created related to this: https://github.com/jmwright/cadquery-freecad-module/issues/144

I need more information.

Are you trying to run this script with Python 3? Are you running your script from within FreeCAD, or on the command line? What version of FreeCAD are you using? You can start the FreeCAD  GUI and go to Help > About FreeCAD

Does "windows also showing error" mean that the error happens in the Windows operating system as well as another? Is the other operating system Linux?

selvakarna, 2019-10-10:
@jmwright ,  Can you share FreeCAD Version with Python Version compatibility for installation, with WINDOWS & Linux OS.

selvakarna, 2019-10-10:
I have installed cadquery  2.3.1 in Linux Machine ?  when i import cadquery its show error ?

Error:

>>> import cadquery
Traceback (most recent call last):
  File "/home/hp/.local/lib/python3.6/site-packages/cadquery/freecad_impl/__init__.py", line 151, in <module>
    import FreeCAD
ModuleNotFoundError: No module named 'FreeCAD'

jmwright, 2019-10-10:
It should work with FreeCAD 0.17, 0.18 and 0.19, and Python 2 and Python 3, but 0.19 is still in development so it's changing, and each OS can act differently.

When you run `which freecad` on your Linux system, what does it show?

selvakarna, 2019-10-11:
Hi @jmwright , when i import cadquery package , i get following error

Python 3.6.8 (default, Aug 20 2019, 17:12:48) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
Traceback (most recent call last):
  File "/home/hp/.local/lib/python3.6/site-packages/cadquery/freecad_impl/__init__.py", line 151, in <module>
    import FreeCAD
ModuleNotFoundError: No module named 'FreeCAD'

During handling of the above exception, another exception occurred:

selvakarna, 2019-10-11:
Hi @jmwright , Can you Please share FreeCAD installation producer PIP python using commands.

jmwright, 2019-10-11:
@selvakarna Assuming you're running Ubuntu, you can follow the instructions [here](https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-stable) to add the FreeCAD stable PPA. Once you're followed the instructions there, you can just run `sudo apt-get install freecad`.

selvakarna, 2019-10-14:
@jmwright , i followed above your instruction , but after that above installation i get some error, please check this below error.

Error:

Python 3.6.8 (default, Aug 20 2019, 17:12:48) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import freecad
>>> import cadquery
Traceback (most recent call last):
  File "/home/hp/.local/lib/python3.6/site-packages/cadquery/freecad_impl/__init__.py", line 151, in <module>
    import FreeCAD
ModuleNotFoundError: No module named 'FreeCAD'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/hp/.local/lib/python3.6/site-packages/cadquery/__init__.py", line 2, in <module>
    from .freecad_impl.geom import Plane,BoundBox,Vector,Matrix,sortWiresByBuildOrder
  File "/home/hp/.local/lib/python3.6/site-packages/cadquery/freecad_impl/__init__.py", line 155, in <module>
    import FreeCAD
ImportError: dynamic module does not define module export function (PyInit_FreeCAD)
>>>

jmwright, 2019-10-29:
@selvakarna If you run `which freecad` from the terminal, what does it show?

closed 2019-11-05 by selvakarna