CadQuery/cadquery-freecad-workbench #145: ModuleNotFoundError: No module named 'FreeCAD'

Zebrafish007, 2020-01-01:
And again the Cluedo question appears on where to find FreeCAD. I've seen a bunch of workarounds for the cadquary __init__.py but this file is since 2017 changed completely.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "Z:\test\CADit.py", line 1, in <module>
    from cadquery import Workplane
  File "x:\y\anac\lib\site-packages\cadquery\__init__.py", line 2, in <module>
    from .freecad_impl.geom import Plane,BoundBox,Vector,Matrix,sortWiresByBuildOrder
  File "x:\y\anac\lib\site-packages\cadquery\freecad_impl\__init__.py", line 153, in <module>
    path = _fc_path()
  File "x:\y\anac\lib\site-packages\cadquery\freecad_impl\__init__.py", line 145, in _fc_path
    raise ImportError('cadquery was unable to determine freecad library path')
ImportError: cadquery was unable to determine freecad library path

The location of FreeCAD: K:\prog\FreeCAD

How to fix this?

Zebrafish007, 2020-01-01:
.. and obviously adding "x:/y/FreeCAD/bin" to the init.py doesn't fix it after adding it here:

    elif sys.platform.startswith('win'):
        # Try all the usual suspects
        for _PATH in [
                "c:/Program Files/FreeCAD0.12/bin",

jmwright, 2020-01-01:
@Zebrafish007 What operating system are you running? If it's Linux, what does `which freecad` show?

jmwright, 2020-01-01:
@Zebrafish007 Nevermind. I wasn't thinking about the drive letters and the `win` in the snippet. Is there still a `bin` directory in FreeCAD on Windows?  The path needs to point to the directory location of the FreeCAD executable.

Zebrafish007, 2020-01-02:
It seems I added the folder in the wrong python environments init.py. Therafter, python3.6.dll error. Created a 36 env. Then a missing pythreejs error. Fixed that. followed by the below error About which I have no clue what that is.

    File "X:\Y\Anac\envs\py36\lib\site-packages\jupyter_cadquery\cad_view.py", line 30, in <module>
        from OCC.Core.Visualization import Tesselator
    ModuleNotFoundError: No module named 'OCC'

jmwright, 2020-01-02:
Your error is related to CadQuery 2.0 (based on PythonOCC) and [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery), is that what you're trying to run? This workbench is only compatible with CadQuery 1.0 (FreeCAD based), and won't work with 2.0. You should be able to run `jupyter notebook` from within the jupyter_cadquery environment you've created, or you could check out [CQ-editor](https://github.com/CadQuery/CQ-editor) if you want a desktop application.

KeithSloan, 2020-01-28:
How feasible is it to update this workbench to work with CQ 2.0 which uses PythonOCC? Clearly you cannot have python that is using PythonOCC outputing directly to a FreeCAD window. I know that Part has  Part.__toPythonOCC__() and Part.__fromPythonOCC__() to exchange TopoDS_Shape entities to/from pythonOCC, would it  be possible to put wrappers round things? I know the answer should be to look at the code, but would like somebody who is already familiar with the code to comment on feasibility of doing this before investigating further.

closed 2021-01-05 by jmwright