CadQuery/cadquery-freecad-workbench #146: Update to support of CG 2.0

KeithSloan, 2020-01-28:
How feasible is it to update this workbench to work with CQ 2.0 which uses PythonOCC? Clearly you cannot have python that is using PythonOCC outputing directly to a FreeCAD window. I know that Part has Part.toPythonOCC() and Part.fromPythonOCC() to exchange TopoDS_Shape entities to/from pythonOCC, would it be possible to put wrappers round things? I know the answer should be to look at the code, but would like somebody who is already familiar with the code to comment on feasibility of doing this before investigating further.

jmwright, 2020-01-28:
@KeithSloan 

## Background/History
When CQ 2.0 was in the earlier stages my plan was to update this workbench, and part of that was going to be using `toPythonOCC` and `fromPythonOCC`. However, as other options became available such as [CQ-editor](https://github.com/CadQuery/CQ-editor) and [jupyter-cadquery](https://github.com/bernhard-42/jupyter-cadquery), and the community didn't seem to be asking for the things this workbench provides, it became harder to justify upgrading this. With that said, I still think good integration and interop with FreeCAD is a good thing.

## Technical
It did not take a lot to update [cadquery-gui](https://github.com/jmwright/cadquery-gui) for CQ 2.0 because CQGI works the same way. I expect that most of the work on this workbench would be done [here](https://github.com/jmwright/cadquery-freecad-module/blob/master/Helpers.py#L4). If you're interested in putting together a PR, it would be nice if the user could select between CQ versions (default to CQ 1.0) instead of being forced to upgrade to CQ 2.0. CQ 2.0 wouldn't be the "batteries included" experience that users of this workbench have come to expect. The settings dialog code is [here](https://github.com/jmwright/cadquery-freecad-module/blob/master/SettingsDialog.py) where a CQ version setting could be added.

KeithSloan, 2020-01-29:
I had a look at "here" and the only thing I saw was
`newFeature.Shape = cqObject.toFreecad()`

So I assume that cg already does some conversion to FreeCAD Shape and I need to look at the cgObject implementation

jmwright, 2020-01-29:
`toFreecad` is from CQ 1.0. There's no equivalent in CQ 2.0. I imagine that code would have to change to something like:
```
newFeature.Shape = Part.fromPythonOCC(cqObject.val().wrapped)
```
But that's just off the top of my head. I also may be missing other changes that will have to be made, but that's where I would start.

infosisio, 2021-05-15:
I was wondering about this thing for a few days before arriving to this thread. CQ-editor may be more than ideal as a primary editor but freecad is a much broader tool, so it makes total sense for CQ to have a shelter there as well.

jmwright, 2021-05-15:
@infosisio I spent some time looking into it, but the main problem I ran into was that FreeCAD would not use an Anaconda environment, so it's very hard to experiment with it. I don't have a lot of time to spend on this ATM, but am supportive of the community working on it. I would like to see CadQuery and FreeCAD integrated again.

infosisio, 2021-05-15:
@jmwright Got it!

grbd, 2022-07-17:
The latest beta version of cadquery now supports being used without conda / with pip
```
pip install cadquery==2.2.0b0
```

jmwright, 2022-08-03:
@grbd Thanks for pointing that out. This main branch needs more than just installing CQ 2.x, and so I have been pointing people to [this fork](https://github.com/jpmlt/freecad-cadquery2-workbench). Tagging @jpmlt to make sure the `pip install` line above is seen for that fork.

jmwright, 2024-12-04:
See #158 (WIP)

jmwright, 2024-12-10:
Addressed in #158

closed 2024-12-10 by jmwright