CadQuery/cadquery-freecad-workbench #147: Examples not running -  module 'time' has no attribute 'clock'

sjdevries, 2020-02-24:
This is running:

OS: Manjaro Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16155 (Git)
Build type: Release
Branch: makepkg
Hash: 9ae129705652eec6be426211e5eaaa16c847b7c5
Python version: 3.8.1
Qt version: 5.14.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```
Executing CQGI-compliant script.
Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "/home/username/.FreeCAD/Mod/cadquery_module/CQGui/Command.py", line 198, in Activated
    build_result = cqModel.build(build_parameters=build_parameters)
  File "/home/username/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/cqgi.py", line 97, in build
    start = time.clock()

module 'time' has no attribute 'clock'
```
Searching brought me to a [Similar Issue](https://stackoverflow.com/questions/58569361/attributeerror-module-time-has-no-attribute-clock-in-python-3-8)

The function time.clock() has been removed, after having been deprecated since Python 3.3: use time.perf_counter() or time.process_time() instead, depending on your requirements.

jmwright, 2020-02-24:
@sjdevries Thanks for reporting this.

jmwright, 2020-02-24:
@sjdevries This is a little tricky because some users are still running on older versions of FreeCAD (and thus older versions of Python), and those two replacement functions were only added in Python 3.3. Until there is a fix in master, you can modify `/home/username/.FreeCAD/Mod/cadquery_module/Libs/cadquery/cadquery/cqgi.py` directly to change the two `time.clock()` calls. If you've installed this workbench via the add-on manager, it might complain when it goes to update the workbench, requiring you to undo your changes to get it to pull the changes. Whether it will complain or not depends on whether the git module is available, or if it does a direct download.

mj1911, 2020-07-13:
As a temporary work-around, `start = time.perf_counter()` (and the end= statement) will get it working in python >= 3.8.

hkienle, 2020-12-31:
Perhaps making the change from `time.clock()` to `time.perf_counter()`could be done after FC 0.19 is out? The FC that I run (0.19.23323) uses Python 3.8.6.

Alternatively, is the built time of much use anyways, maybe just remove it!?

jmwright, 2020-12-31:
That time call is actually in the old CadQuery repo, so it has to be changed there. However, that repo has been left behind as development has moved [here](https://github.com/CadQuery/cadquery). I'm leaving this issue open so in the hopes that I'll have time to circle back and fix some of these problems, but in general, I encourage users to move to CadQuery 2.x and [CQ-editor](https://github.com/CadQuery/CQ-editor). I think that most users have migrated at this point.

hkienle, 2021-01-02:
Thanks for clarification. I came from the "FreeCAD world", looking for interesting workbenches and hit on CadQuery this way. I am interested at this point to see how I can use CadQuery within the FC context, this is perhaps untypical.

jmwright, 2021-01-02:
@jmwright I would say it's not typical to use this workbench for CadQuery work now, but you can use it if you want to. However, CadQuery has changed a lot since since this workbench was the standard, and it's non-trivial to make CadQuery 2.x work within it. If you use this workbench you'll be missing out on a lot of features and fixes that you will get if you use CadQuery 2.x and CQ-editor.

jmwright, 2021-01-05:
This has been fixed in commit ac4cbd5e96f8707f30cf226e2be861013be83823

closed 2021-01-05 by jmwright