CadQuery/cadquery-freecad-workbench #148: show() not defined

hkienle, 2020-12-31:
The show() function is mentioned, for example in
- script_template.py
- Ex001_Simple_Block.py

but using it results in
`Error executing CQGI-compliant script. name 'show' is not defined
`
Perhaps this should be updated.

jmwright, 2020-12-31:
Hi @hkienle
Unless you have a compelling reason to use CadQuery from FreeCAD, I would encourage you to check out the latest version of CadQuery [here](https://github.com/CadQuery/cadquery) and [CQ-editor](https://github.com/CadQuery/CQ-editor) to use as a GUI for writing and displaying the results of CadQuery scripts. This module does not get much use these days because it's not compatible with the current version of CadQuery.

As for your original issue, the examples have been moved to the core CadQuery repo and updated there to use `show_object()`. The example in this repo is getting pretty old and I missed converting `show()` to `show_object()`. Thanks for reporting it.

hkienle, 2021-01-02:
Thanks for the pointers. Before your post, I found only the CQ-editor repository and missed the documentation given at the cadquery repo.

As you might know, the CadQuery workbench has a page at the FreeCAD wiki: https://wiki.freecadweb.org/CadQuery_Workbench

I would like to update it to the current status. Is is fair to say?:
- The active development of CadQuery has moved away from FreeCAD [https://github.com/CadQuery/cadquery#why-this-fork], offering now a stand-alone GUI [https://github.com/CadQuery/CQ-editor].
- The CadQuery workbench is no longer actively maintained, but may receive sporadic fixes by its author.

Of course, if you prefer to update the FreeCAD wiki yourself that would be even better!

jmwright, 2021-01-02:
@hkienle If you could update the wiki that would be great. Otherwise I'll get to it when I can. I think what you propose sounds great, and reflects the current state accurately.

hkienle, 2021-01-04:
Thanks for your OK, Wiki is updated: https://wiki.freecadweb.org/CadQuery_Workbench

jmwright, 2021-01-04:
Thanks for doing that @hkienle

hkienle, 2021-01-07:
@jmwright Thank you so much for making the changes, especially since the focus is now on CQ 2.0!