CadQuery/cadquery-freecad-workbench #150: Cannot import name 'Assembly' from 'cadquery'

davidvpspace, 2021-04-13:
Hi there,

I am currently working with the Cadquery module on FreeCAD. I am able to create parts without problems, but when I try to create the assembly, it seems like cadquery does not have the class Assembly. I am trying a simple code like:

`import cadquery as cq
from cadquery import Assembly`

and I get the following error:

`Running the Python command 'CadQueryExecuteScript' failed:
Traceback (most recent call last):
  File "C:\Users\david\AppData\Roaming\FreeCAD\Mod\cadquery_module\CQGui\Command.py", line 236, in Activated
    imp.load_source('__cq_freecad_module__', tempFile.name)
  File "C:\Program Files\FreeCAD 0.19\bin\lib\imp.py", line 171, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\david\AppData\Local\Temp\tmpbkygo96y", line 4, in <module>
    from cadquery import Assembly
cannot import name 'Assembly' from 'cadquery' (C:\Users\david\AppData\Roaming\FreeCAD\Mod\cadquery_module\Libs\cadquery\cadquery\__init__.py)`

I did not do anything special, I've just intalled cadquery from freeCAD using the pluggin manager. I have tried to ininstall and install but same problem arises.

jmwright, 2021-04-13:
Hi @davidvpspace This workbench is not compatible with CadQuery 2.x at this time, and Assemblies are a CadQuery 2.x feature. CQ-editor is the recommended GUI for CadQuery 2.x.

Is there a reason you're using this FreeCAD workbench? Some people have workflows that they want to keep all in FreeCAD, but if you're able to switch to CQ-editor, that will give you the best support right now.

davidvpspace, 2021-04-13:
Thanks @jmwright ! I am a regular user of FreeCAD that's why I was using it instead of the CadQuery GUI. I have installed it and I can properly work now with assemblies. Thank you for you help. 

One more question if I may, is it possible to export an assembly as an .stl file?

Thanks again

jmwright, 2021-04-13:
> is it possible to export an assembly as an .stl file?

Not at this time. Assembly [save](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=save#cadquery.Assembly.save) only supports STEP and OCCT's native XML format.

This work-around might help you though: https://github.com/CadQuery/cadquery/issues/629#issuecomment-794389876

davidvpspace, 2021-04-13:
Thanks a lot!

closed 2021-04-13 by davidvpspace