CadQuery/cadquery-freecad-workbench #151: No module named 'ExportCQ'

infosisio, 2021-06-07:
Tried installing the latest release with the latest version of freecad and this is what I get:

```
During initialization the error No module named 'ExportCQ' occurred in /usr/share/freecad/Mod/cadquery-freecad-module/InitGui.py
Please look into the log file for further information
``
I went through the installation steps.
Am I missing something?

jmwright, 2021-06-07:
Are you trying to use this FreeCAD module with CadQuery 2.x? It's only compatible with CadQuery 1.x at this time.

infosisio, 2021-06-07:
@jmwright No, CadQuery 1.x. I just followed the instructions on the installation page, I thought the right CQ version was self contained. Do I need to install anything? If so, could you please redirect me to the right page?

infosisio, 2021-06-07:
@jmwright No, cadquery 1.x but I thought it was self contained. I installed cadquery with pip though, it still won't work.

jmwright, 2021-06-07:
Which version of FreeCAD are you running, and on which OS?

infosisio, 2021-06-07:
Freecad 0.18.4, Ubuntu 20.04.2 LTS

jmwright, 2021-06-07:
Did you build FreeCAD from source or anything? The master branch hasn't changed in 5 months, and I haven't had this error reported before that I remember.

infosisio, 2021-06-07:
@jmwright Yeah, I re-installed Freecad to make sure, same error.

infosisio, 2021-06-12:
@jmwright After changing something system-wide, I can now confirm that cadquery workbench works.

closed 2021-06-12 by infosisio