CadQuery/cadquery-freecad-workbench #152: Rotation of assembly or imported step

davidvpspace, 2022-01-04:
Hello!

I have an simple assembly created in cadquery (assy). I would like to add an imported STEP file (frame) and add it to an assembly containing the previous and the step file. However, I would like to either rotate the step file or the assembly before putting them together.

I want to rotate the step file so that afterwards I can place it to the desire position with respect to the assembly using cq.Location() inside the add method. This is an example of my code:

```
frame = cq.importers.importStep("frame.stp")  ##import step file
frame.rotateAboutCenter((1,0,0), 90.0)  ##rotate about X axis 90 deg
assy.add(self.frame,  loc=cq.Location(cq.Vector(-50,50,100))) ##add step to assembly and place to desired location
```

The code gives no error, but nothing happens. Position of the imported step is the same. I have tried changing the rotation direction and rotation angle, but it's like rotateAboutCenter doesn't do anything to the steo file. 

Waht would be the correct way to rotate an imported step? For my intentions, it would be also possible to rotate the assembly before adding the steo file to it.

Thanks in advance!

jmwright, 2022-01-04:
This workbench only supports CadQuery 1.x, and you need a newer version of CadQuery 2.x for the assembly functionality. If you want to use CadQuery within FreeCAD I would recommend trying [this fork](https://github.com/jpmlt/freecad-cadquery2-workbench) of this workbench, which supports CadQuery 2.x (installed separately). Otherwise you can look into [CQ-editor](https://github.com/CadQuery/CQ-editor).

If you are just looking for general CadQuery help, you can open an issue on that repo or check out the Google Group or Discord. https://github.com/CadQuery/cadquery#getting-help

davidvpspace, 2022-01-04:
Alright! Thanks

closed 2022-01-04 by davidvpspace