CadQuery/cadquery-freecad-workbench #153: module 'collections' has no attribute 'MutableMapping'

scrouthtv, 2022-01-30:
Hello,

I tried installing cadquery through the Addon Manager in FreeCad. However, when starting FreeCad, I get this error in the log, and the workbench is not available:
```
During initialization the error "module 'collections' has no attribute 'MutableMapping'" occurred in /home/lenni/.FreeCAD/Mod/cadquery-freecad-module/InitGui.py
Please look into the log file for further information
```

I am using python 3.10.1

scrouthtv, 2022-01-30:
I fixed it for now by deleting the bundled pyparsing and using my OS's pyparsing 2.4.7, however I think this needs to be addressed.

jmwright, 2022-01-30:
This repo is only compatible with CadQuery 1.x and is not actively maintained anymore. However, there is a fork of this workbench (link in readme) that does support CadQuery 2.x. I would recommend running that.

scrouthtv, 2022-01-31:
I am aware of that, and need CadQuery 1.x for KiCad 3D model generation.

jmwright, 2022-01-31:
The model generation in KiCAD is being updated to use CadQuery 2.x. My understanding is that the plan is to generate the models in GitLab's CI system at first and bundle them with KiCAD. After that the plan is to embed the CadQuery scripts with KiCAD and generate the models on demand. My fork of the generator repo where I am doing the update in collaboration with the KiCAD project is [here](https://gitlab.com/jmwright/kicad-packages3D-generator). An example of a library that's been converted already can be found [here](https://gitlab.com/jmwright/kicad-packages3D-generator/-/tree/master/Connector_Dsub).

The general workflow is this:

* Create a cq_parameters.yaml file holding the parameters for each component variant
* Create your module that generates the model's parts in CadQuery
* Create the main_generator.py file (best to copy from an existing conversion directory and edit it)
  * Have this script import your module and call the relevant methods to get your CadQuery model objects
  * Plug the objects into the assembly that is in main_generator.py, making sure that the correct colors are being loaded for each part of the package

This method allows the models to be generated without any kind of GUI, and the assembly allows direct STEP and VRML export. There is still an issue with the VRML files, but I think it may be due to a difference between the embedded version of the OpenCASCADE kernels between CadQuery and KiCAD.

So I would still encourage you to use something like CQ-editor with CadQuery 2.x instead of this workbench to create your models, then have the main_generator script import your model generation module. If you use this workbench and CadQuery 1.x, the library you create will have to be updated right away.

scrouthtv, 2022-01-31:
For reference: https://gitlab.com/kicad/libraries/kicad-packages3D-generator/-/merge_requests/15

I am definitely not touching the GW generator (for now)...

jmwright, 2022-01-31:
Ah, I see. You're wanting to test the parameter updates in your merge request. It sounds like you have the workbench working now, correct?

easyw, 2022-09-23:
@jmwright 
I've done a PR here https://github.com/jmwright/cadquery-freecad-module/pull/156
to fix py3.10 compat

jmwright, 2022-09-23:
Big thanks to @easyw for fixing this.

@scrouthtv Please re-open if there is an aspect of this issue that is not addressed by #156 

@easyw Is this FreeCAD workbench still relevant with the work that has been done to upgrade the KiCAD model generation infrastructure to CadQuery 2.x? There is a fork of this workbench that is compatible with CadQuery 2.x, but it takes some manual install steps.

easyw, 2022-09-24:
> Is this FreeCAD workbench still relevant with the work that has been done to upgrade the KiCAD model generation infrastructure to CadQuery 2.x?

IMO this WB is very useful because it uses the FC GUI... it was very easy, during the developing of the 3D lib, having the ability to show intermediate state of the cq 3D result.

> There is a fork of this workbench that is compatible with CadQuery 2.x, but it takes some manual install steps.

I have noticed this only few times ago... what I'm missing ATM is that the new release is 'Assembling' the models instead of generating a single unioned 3D model, as it was i the old libraries.
i.e. a LQFP 144pin will be a nested object with the body the body marker and 144 pins...
is there an option to generate te models as it was in the original libraries?

easyw, 2022-09-25:
@jmwright 
do you know if it is possible to use CQ in MSYS2?

easyw, 2022-09-25:
I used pip to install cadquery, but then I go this error:
```
 in <module>
    import cadquery as cq
  File "C:/msys64/mingw64/lib/python3.10/site-packages/cadquery/__init__.py", line 6, in <module>
    from .freecad_impl.geom import Plane,BoundBox,Vector,Matrix,sortWiresByBuildOrder
  File "C:/msys64/mingw64/lib/python3.10/site-packages/cadquery/freecad_impl/__init__.py", line 154, in <module>
    path = _fc_path()
  File "C:/msys64/mingw64/lib/python3.10/site-packages/cadquery/freecad_impl/__init__.py", line 146, in _fc_path
    raise ImportError('cadquery was unable to determine freecad library path')
ImportError: cadquery was unable to determine freecad library path
```
does cq relys on freecad?

jmwright, 2022-09-26:
> is there an option to generate the models as it was in the original libraries?

Not at this time, although you can call `.toCompound()` on the assembly to create a single compound model. However, you lose the color data. That's why I used the assembly method.

> does cq relys on freecad?

Only CQ 1.x relies on FreeCAD. If you simply ran `pip install cadquery`, then you got the old version. You currently need to install a pre-release (`pip install --pre cadquery`) or from git to have an up-to-date install of CQ 2.x. `pip install git+https://github.com/CadQuery/cadquery.git`.

> do you know if it is possible to use CQ in MSYS2?

I don't have any experience with CQ in MSYS2.

closed 2022-09-26 by jmwright

easyw, 2022-09-26:
> > is there an option to generate the models as it was in the original libraries?
> 
> Not at this time, although you can call `.toCompound()` on the assembly to create a single compound model. However, you lose the color data. That's why I used the assembly method.

that is a big issue in generating the 3d library... do you think it could be implemented?

> Only CQ 1.x relies on FreeCAD. If you simply ran `pip install cadquery`, then you got the old version. You currently need to install a pre-release (`pip install --pre cadquery`) or from git to have an up-to-date install of CQ 2.x. `pip install git+https://github.com/CadQuery/cadquery.git`.

Thanks, I'm going to try it... I did ...
It seems it miss cq-ocp
`No matching distribution found for cadquery-ocp`
I then tried:
`pip install git+https://github.com/CadQuery/OCP.git`
and I had:
`ERROR: git+https://github.com/CadQuery/OCP.git does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.`

jmwright, 2022-09-26:
> that is a big issue in generating the 3d library... do you think it could be implemented?

@easyw I don't really understand what the problem is. Could you open an issue on the generator repo with an explanation of why the current assembly setup won't work, and whether it applies to STEP, VRML, or both? https://gitlab.com/kicad/libraries/kicad-packages3D-generator

> No matching distribution found for cadquery-ocp

What OS are you running FreeCAD on, and what version of Python is bundled?

easyw, 2022-09-26:
> I don't really understand what the problem is. Could you open an issue on the generator repo with an explanation of why the current assembly setup won't work, and whether it applies to STEP, VRML, or both?

I will do it...

> What OS are you running FreeCAD on, and what version of Python is bundled?

I'm building KiCAD with MSYS2 (it should be python 3.10.6)
https://dev-docs.kicad.org/en/build/windows-msys2/

easyw, 2022-09-27:
@jmwright 
done here https://gitlab.com/kicad/libraries/kicad-packages3D-generator/-/issues/5

jmwright, 2022-09-27:
> I'm building KiCAD with MSYS2 (it is python 3.10.6)

The wheel builds on PyPi only cover Python 3.8 through 3.10 (so you're fine there), and OSes Linux x86_64, Windows 64-bit and MacOS (Intel based only, I think). I wonder if MSYS2 is making your system appear as an OS and/or an architecture that is not in that list. You could try retrieving the system info that Python sees to check if that has a clue in it.