CadQuery/cadquery-freecad-workbench #157: Embedded libs are breaking Plot module again

hasecilu, 2024-10-23:
Similarly as I commented on https://github.com/jmwright/cadquery-freecad-module/pull/154#issuecomment-1894439763 the embedded libraries conflict with the new versions that should be used on FreeCAD and break plot capabilities

Just commenting this lines makes, plot work again. Alternatively also deleting this WB fixes the problem.
```python
# sys.path.insert(0, libs_dir_path)
```

I think would be nice to prepare this WB for 1.0 making it to use system libraries.

<https://wiki.freecad.org/Package_Metadata>

I think it nwould be something lika this:
```xml
  <content>
    <workbench>
      <classname>CadQueryWorkbench</classname>
      <subdirectory>./</subdirectory>
      <icon>/CQGui/Resources/icons/CQ_Logo.svg</icon>
      <depend type="python">fpdf2</depend>
      <depend type="python">certifi</depend>
      <depend type="python">chardet</depend>
      <depend type="python">docutils</depend>
      <depend type="python">future</depend>
      <depend type="python">idna</depend>
      <depend type="python">jinja2</depend>
      <depend type="python">markupsafe</depend>
      <depend type="python">pint</depend>
      <depend type="python">requests</depend>
      <depend type="python">tinydb</depend>
      <depend type="python">urllib3</depend>
    </workbench>
  </content>
```

hasecilu, 2024-10-29:
Yes, in the workbenches I have done I have not needed to do it (add dependencies) but this WB seems to need it. There are examples files on the Wiki.

Also you may need to send a patch to FreeCAD addon repo to add the packages to the allowedlist: https://github.com/FreeCAD/FreeCAD-addons/blob/master/ALLOWED_PYTHON_PACKAGES.txt

In that case you can consult Chris H. there, he knows more about it.

jmwright, 2024-12-04:
Big changes in draft PR #158 includes a minimal [package.xml](https://github.com/jmwright/cadquery-freecad-module/pull/158/files#diff-37a67ff78eb7260214b323353263b9af40a2fa98719a1e81937fae0159df87a3) file. The workbench as-is in the PR is not ready to use yet, but results so far are promising. It is not clear to me though how I test installation of a workbench locally before requesting that it be updated in the workbench repo/wiki.

hasecilu, 2024-12-04:
You can add a custom URL of the repo on `Edit > Preferences > Addon Manager > Custom repositories` you can enter a specific branch also.

jmwright, 2024-12-10:
Addressed in #158

closed 2024-12-10 by jmwright