CadQuery/cadquery-freecad-workbench #161: cadquery mod not working with Freecad 1.0.0 appimage

benjinne, 2024-12-31:
Hi, 

Whenever I try to install this mod through the Freecad addon manager: I get this error:
[image]

Then after restarting, and trying to use `import cadquery` in the python terminal, I get module not found.
If I then try to install it using `subprocess.run(["python", "-m", "pip", "install", "--upgrade", "git+https://github.com/CadQuery/cadquery.git"], capture_output=True)`

It seems to install, however now I'm getting this error when trying to import cadquery:

```
import cadquery
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/tmp/.mount_FreeCAgIbCng/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ben/.local/lib/python3.11/site-packages/cadquery/__init__.py", line 10, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/tmp/.mount_FreeCAgIbCng/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ben/.local/lib/python3.11/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
  File "/tmp/.mount_FreeCAgIbCng/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ben/.local/lib/python3.11/site-packages/OCP/__init__.py", line 14, in <module>
    from OCP.OCP import *
  File "/tmp/.mount_FreeCAgIbCng/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: libvtkWrappingPythonCore3.11-9.2.so: cannot open shared object file: No such file or directory
```

My freecad details:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * cadquery-freecad-workbench-2.0.0 2.0.0
  * kicadStepUpMod 11.4.4
```

jmwright, 2024-12-31:
@benjinne Thanks for trying this update out. 

As for the installation error during the automated install, I think we'll have to wait until CadQuery 2.5 is released because 2.5 has changes needed to make it play better with FreeCAD, and FreeCAD's addon manager errors out on the git repo URL.

The second error is one that I got earlier today, and is related to a beta package (7.7.2.2b2) of cadquery_ocp. You can work around this for now by installing cadquery_ocp==7.7.2 via the subprocess.run method you show above. Then restart FreeCAD and it should work.

```python
subprocess.run(["python", "-m", "pip", "install", "--upgrade", "cadquery_ocp==7.7.2"], capture_output=True)
```

Once that is done and you have activated the CadQuery workbench, you should be able to run the following script in the macro editor and have a document tab open with the 3D box in it.

```python
import cadquery as cq

box = cq.Workplane().box(10, 10, 10)
box.val().label = "BOX"

print("Rendering BOX")

show_object(box, options={"color": (64, 164, 223), "alpha": 0.5})
```

**Please note**: This update was the bare minimum to get CadQuery working with FreeCAD again. I also noticed that you have the `kicadStepUpMod` installed. It will probably take some effort to get this workbench and that module working together again. However, I do not really know what would be involved in fixing compatiblility.

benjinne, 2024-12-31:
Thanks!
`subprocess.run(["python", "-m", "pip", "install", "--upgrade", "cadquery_ocp==7.7.2"], capture_output=True)`
fixed it for me.

benjinne, 2025-01-01:
@jmwright I'm running into a problem when trying to execute a python script from outside the appimage.
For example:

~/Desktop/FreeCAD_1.0.0-conda-Linux-x86_64-py311.AppImage test.py

With the contents of test.py being:
```
import cadquery as cq
box = cq.Workplane().box(10, 10, 5)
box.val().label = "My_Box"
show_object(box)
```

I keep getting
`Exception while processing file: test.py [name 'show_object' is not defined]`

I think it's because the cadquery module isn't being referenced from my machine and not the appimage
Any idea if it's possible to execute a script outside of the macro menu?

jmwright, 2025-01-01:
I would guess it's because `show_object` isn't injected into the globals until the workbench is activated.

https://github.com/CadQuery/cadquery-freecad-workbench/blob/master/InitGui.py#L93

CadQuery doesn't define `show_object`, each execution environment is expected to do that. I'm not sure how to make it available in this case.

jmwright, 2025-01-01:
Maybe I could put the `show_object` declaration in `Init.py` instead, but that method really needs the GUI to be inited to work.

benjinne, 2025-01-01:
> Maybe I could put the `show_object` declaration in `Init.py` instead, but that method really needs the GUI to be inited to work.

It seems like it used to work or maybe with windows versions:
I'm trying to something similar to this: https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/jst#running-the-scripts

benjinne, 2025-01-01:
> It seems like it used to work or maybe with windows versions: I'm trying to something similar to this: [...]

I found a solution:

```
import time
import FreeCAD as App
import FreeCADGui as Gui
# Activate the desired workbench
Gui.activateWorkbench("CadQueryWorkbench")

# Wait until the workbench is fully activated
timeout = 5  # Timeout in seconds
start_time = time.time()
while Gui.activeWorkbench().name() != "CadQueryWorkbench":
    print(str(Gui.activeWorkbench()))
    if time.time() - start_time > timeout:
        raise TimeoutError("Timed out waiting for CadQueryWorkbench to activate.")
    time.sleep(1)

# Proceed with your script
print("CadQueryWorkbench is active!")

import cadquery as cq
import os
print(os.path.dirname(cq.__file__))

FreeCAD.Console.PrintMessage('\r\nRunning...\r\n')

box = cq.Workplane().box(10, 10, 5)
box.val().label = "My_Box"
show_object(box)

FreeCAD.Console.PrintMessage('\n\nDone\n')
```

jmwright, 2025-01-01:
@benjinne I re-added `Init.py` and put the `show_object` injection in there, and the script runs for me now from the command line without any errors, and without the need for timers. Please update and let me know if it works for you.

benjinne, 2025-01-02:
> @benjinne I re-added `Init.py` and put the `show_object` injection in there, and the script runs for me now from the command line without any errors, and without the need for timers. Please update and let me know if it works for you.

Hey that works better now and really simplifies my previous example into:
```
import cadquery as cq
box = cq.Workplane().box(10, 10, 5)
box.val().label = "My_Box"
show_object(box)
```

However, I found an interesting bug that causes the show_object function to dissapear:
When I run:
```

from PySide import QtGui
import FreeCADGui as Gui
import cadquery as cq
Gui.activateWorkbench("CadQueryWorkbench")
try:
    box = cq.Workplane().box(10, 10, 5)
    box.val().label = "My_Box"
    show_object(box)
except Exception as e: # catch *all* exceptions
    print(e)
```

I get: `name 'show_object' is not defined`

I tracked it down to `from PySide import QtGui` which does something weird to cause show_object to be unavailable. Another weird problem is when I move the code outside the try except, it works just fine. 

Is there a way to import the show_object function directly instead of relying on the global?

jmwright, 2025-01-02:
I pushed another update which breaks `show_object` out into a separate `display.py` module so that it can be imported. The following code works fine for me now when run by FreeCAD from the command line. You do not need to activate the CadQuery workbench unless you need to for some other purpose.

```python
# from PySide import QtGui
# import FreeCADGui as Gui
from CQGui.display import show_object
import cadquery as cq
# Gui.activateWorkbench("CadQueryWorkbench")
try:
    box = cq.Workplane().box(10, 10, 5)
    box.val().label = "My_Box"
    show_object(box)
except Exception as e: # catch *all* exceptions
    print(e)
```

This is not a great solution IMO because I would like for scripts to be portable between CQ-editor and FreeCAD without modification. I am not sure what is going on that causes `show_object` to be lost when running from within a try/except block though (maybe something with scope?), and this method allows `show_object` to be used no matter what else happens during initialization.

The following code also works, just by moving the `show_object` call outside of the try/except block, which makes me think that the problem is a scope thing.

```python
#from PySide import QtGui
#import FreeCADGui as Gui
#from CQGui.display import show_object
import cadquery as cq
#Gui.activateWorkbench("CadQueryWorkbench")
box = None
try:
    box = cq.Workplane().box(10, 10, 5)
    box.val().label = "My_Box"
except Exception as e: # catch *all* exceptions
    print(e)

show_object(box)
```

benjinne, 2025-01-02:
Nice! that works, however:
`from CQGui.display import show_object`
Causes this error `Tried to run Gui::Application::initApplication() twice!`
It doesn't seem to break anything, but just something I noticed

jmwright, 2025-01-02:
@benjinne I changed the scope of the imports so that they do not happen at the top level of the display module, and that has fixed the double initialization problem. Please update and have a look.

benjinne, 2025-01-02:
Fixed!

jmwright, 2025-01-02:
@benjinne I am not excited about how this works, but FreeCAD's addon manager will not install CadQuery's dependencies automatically (even now that 2.5.0 has been released), so I had to add a fallback install in `Init.py`.

https://github.com/CadQuery/cadquery-freecad-workbench/blob/master/Init.py#L7

I get a little bit concerned whenever I start messing with the environment, but I am not sure what else to do. It's possible that I could wait until the workbench loads and then present the user with a Yes/No dialog before installing. Thoughts?

benjinne, 2025-01-02:
> @benjinne I am not excited about how this works, but FreeCAD's addon manager will not install CadQuery's dependencies automatically (even now that 2.5.0 has been released), so I had to add a [...]

I don't have any good solutions, but I did notice there's a python dependencies menu that doesn't seem used. Maybe there's a way to use it?
[image]

jmwright, 2025-01-03:
> I did notice there's a python dependencies menu that doesn't seem used. Maybe there's a way to use it?

I have not been able to get the dependencies from this workbench's `package.xml` to show up in that listing. I have been over the package specification's `depend` entry several times and have tried different things, but cannot get it to work. I guess about the only thing I can do is ask on the forum.

jmwright, 2025-01-05:
@benjinne I'm going to close this issue and open a new one for the remaining installation issue.

I did add new manual installation options to the `CadQuery` menu, and created documentation about that method [here](https://github.com/CadQuery/cadquery-freecad-workbench/blob/master/docs/installation.md#install-cadquery-stable).

closed 2025-01-05 by jmwright