CadQuery/cadquery-freecad-workbench #163: updating existing document not working in Freecad 1.0.0

benjinne, 2025-01-02:
https://github.com/CadQuery/cadquery-freecad-workbench/blob/52e7ce166639b09b9cd61a52fd74e625190f01ac/CQGui/display.py#L18

If I test: 
`doc_name mdi_area.activeSubWindow().windowTitle().split(".py")[0]`
then
`FreeCAD.getDocument(doc_name)`
 is never true

doc_name seems to always add a weird `: 1[*]` suffix and it causes getDocument to never recognize it.
I tried saving the document and it still doesn't work.

I think this means calling `show_object()` won't update the existing object, rather create a new one every time.

jmwright, 2025-01-02:
@benjinne I cannot reproduce this. Below, I changed one of the dimensions of the box and re-executed, and the correct document tab was updated.

[image]

[image]

jmwright, 2025-01-02:
@benjinne Which operating system are you running? I have only tested this workbench on Linux since that is my primary OS, so maybe there is a platform difference?

benjinne, 2025-01-02:
> @benjinne Which operating system are you running? I have only tested this workbench on Linux since that is my primary OS, so maybe there is a platform difference?

I think I hit some odd edgecase. I'm working on updating this script: https://github.com/easyw/kicad-3d-models-in-freecad/blob/3618936b169f0b7ddc1d34f50f38e976a95e6e7b/cadquery/FCAD_script_generator/jst/main_generator.py#L174

It seems to work now if I set the model name before show_object

```
    for i in range(len(cq_obj_data)):
        cq_obj_data[i].val().label = ModelName + obj_suffixes[i]
        color_i = colors[i] + (0,)
        show_object(cq_obj_data[i], options={"color": color_i, "alpha": 0.0})
```

If I remove setting the label before show_object it causes this odd behavior

jmwright, 2025-01-02:
So for your PR, do you want to check to see if a `:` is in the window title first, and then implement your split on `:` logic so that we are covered for this edge case? If you do not check for `:` first, I am worried that it will break non-edge cases.

benjinne, 2025-01-02:
> So for your PR, do you want to check to see if a `:` is in the window title first, and then implement your split on `:` logic so that we are covered for this edge case? If you do not check for `:` first, I am worried that it will break non-edge cases.

I'm not sure I like this approach either. If a file has a colon it'll strip everything before it.
If the label is set it should work so I'll probably end up closing the PR.

benjinne, 2025-01-04:
> So for your PR, do you want to check to see if a `:` is in the window title first, and then implement your split on `:` logic so that we are covered for this edge case? If you do not check for `:` first, I am worried that it will break non-edge cases.

Hey @jmwright I mistakenly thought setting the label fixed my issue, but I forgot to revert my changes here. The issue is easy to replicate if you simply close the python script, and run it via Macro->recent macros. Every time you run it, it'll create a new window with the odd glitch. I think my PR should still be used.

jmwright, 2025-01-04:
Ok, PR merged, thanks!

PR #164 (2025-01-04): fix show_object updating the same object
fixes: #163
closed 2025-01-04 by jmwright