CadQuery/cadquery-freecad-workbench #165: cadquery and cadquery-ocp Dependencies Will Not Install Through Addon Manager

jmwright, 2025-01-05:
I keep getting an error when trying to install this workbench through the addon manager, although the workbench will still install. I have tried several things without success. I have posted a question about it on the FreeCAD forum [here](https://forum.freecad.org/viewtopic.php?t=93604). Hopefully I will get an answer.

In the meantime, I have added manual installation options in the `CadQuery` menu, documented [here](https://github.com/CadQuery/cadquery-freecad-workbench/blob/master/docs/installation.md#install-cadquery-stable).

ab12gu, 2025-01-06:
Hey @jmwright i got the same error as u posted in the forum. But also couldn't get the manual installation to work? Like i move it to mod folder but in freecad I don't know how to make and run a script?? were you able to make scripts?

will try 1.0 shortly through manual install

jmwright, 2025-01-07:
@ab12gu I'm glad you got it working. As you discovered, that video is very out-of-date.

You can still name the script with a .py extension even though it is in the macro editor, and you can also start FreeCAD in the terminal and pass it the path to a CadQuery script and it should render the CadQuery model.

The goal with this reboot of the workbench is to fit as much into the standard FreeCAD workflow as possible. That meant using the built-in macro editor, but I also hope to add a file watcher that allows people to use an external editor too (i.e. VSCode).

jmwright, 2025-01-12:
According to a response from ickby on the forum, Python packages have to be added to a whitelist in FreeCAD's repository. So it's certainly not a quick fix. I will plan to make the installation options from the menu more prevalent in the readme.