CadQuery/cadquery-freecad-workbench #166: How to select build123d after installing when running macro?

ab12gu, 2025-01-07:
[image]

I can run a macro, but how do I select to use build123d over cadquery?

jmwright, 2025-01-07:
@ab12gu Please update the workbench and try again. The following example shows all the cases that should work now. Thanks to @jdegenstein for the help on this.

```python
from build123d import *

a = Box(3,2,1)
a.label = "BoxyBox"
with BuildPart() as p:
  Box(3,2,3)
show_object(a) # should work
#show_object(p.part) # should work
#show_object(p) # should work
```

jmwright, 2025-01-07:
Based on a conversation on Discord this works, so I'm closing this issue.

closed 2025-01-07 by jmwright

jdegenstein, 2025-01-07:
Thanks @jmwright for adding these changes to allow build123d to work in this workbench!